chipsalliance/UHDM-integration-tests #25: Output HTML-rendered UHDM progress output to github.io as part of CI

hzeller, 2020-06-18:
The Surelog tests generate `surelog.uhdm.chk.html` files that render input files with annotations about parts that are not covered yet. This helps estimate progress and drill down particular parts.
These are generated as part of running the Surelog tests.

It would be good to have these HTML files available as part of the CI output, copied to github.io pages for easy linking (similar to how we generate the [sv-tests](https://symbiflow.github.io/sv-tests/) output).

@alaindargelas is the surelog.uhdm.chk.html the only one I need to look out for ? IIRC you also had an overview list that shows the coverage percentages.

hzeller, 2020-06-18:
Looks like I don't have permission to assign myself to the bug, can you do that ?

alaindargelas, 2020-06-18:
Hi Henner,
I added you in the group.

The html files contain % so that is all you need. You also have to bring
all the detailed chk<n>.html files that reside next to it along with the
main one for detailed source navigation.