chipsalliance/UHDM-integration-tests #49: Yosys uhdm frontend fails on parsing 2D arrays in module IOs

kamilrakoczy, 2020-10-30:
Yosys uhdm frontend fails on parsing 2D arrays in module IOs. It is creating AST node, that can't by simplified to AST_CONSTANT. It makes yosys to fail on detecting width with message:
```
Unsupported expression on dynamic range select on signal
```

kbieganski, 2020-11-03:
This error is caused by Yosys not finding necessary information about wires, because the module is incomplete (it's a parametrized module that's unused, so Surelog doesn't fully populate it). We should detect these incomplete modules and remove them from the design.

kbieganski, 2020-11-03:
Should be fixed by https://github.com/antmicro/yosys/commit/1b3be994f2efadf5222d93638608574ab190cdcf

kbieganski, 2020-11-06:
Merged by @kamilrakoczy

closed 2020-11-06 by kbieganski