chipsalliance/UHDM-integration-tests #51: Yosys frontend wrongly parses assignment-patters in parameters

kamilrakoczy, 2020-10-30:
Assignment-patterns AST differs when used in parameters parameters (there is no AST_ASSIGN when assigning to parameter). Currently yosys frontend fails on parsing this type of parameters and creates parameter without any children. This ends up with segmentation fault.

kbieganski, 2020-11-03:
Should work for at least some cases after this:
https://github.com/antmicro/yosys/commit/a97a177f9cf1dff95a73c4b5b9ee9d0f743992c3#diff-974349ca5d1dd48f4fbe86dd58c4908830990d8a2272ba93c9317b4de512f78a

kamilrakoczy, 2020-11-03:
Yes, this commit fixes some cases of segmentation fault, but there are also other cases that causes segmentation fault  when parsing parameters (not necessary related to assignment patterns).

PR #56 (2020-11-05): Remove prim_lfsr.sv from skip list
Fixes #51
Fixes #52
closed 2020-11-05 by kamilrakoczy