chipsalliance/UHDM-integration-tests #78: Yosys regressions with Surelog a2e1861

kbieganski, 2020-12-01:
There are some problems in Yosys that appeared with the newest Surelog:
- missing handling of typespec 618 (`vpiStructVar`),
- a segfault in the cleanup procedure after parsing Ibex,
- a segfault after parsing Ibex and passing the AST on to `simplify`,
- tests `IndexedPartSelect`, `PartSelect`, `VarSelect` are failing.

I suspect at least a few of these share their root cause.

kbieganski, 2020-12-02:
Most of the problems were caused by a hex string parsing function in Yosys silently failing.
I fixed the tests and segfaults with this: https://github.com/antmicro/yosys/commit/6a1a853ea2ef5af364f22ecf1cf2f28e4a289e1c

But now the Ibex bitstream isn't working on hardware anymore. I narrowed down the Surelog commit that caused this to https://github.com/alainmarcel/Surelog/commit/241bf562ca31e87482b2cd9f3c31f5342ef37867

Looking at the changes, and the differences in the UHDM dump, it seems to me like changes to parameter handling in Surelog are a likely cause of this.

kamilrakoczy, 2020-12-03:
The remaining problem was is bin string parsing, it is fixed with: https://github.com/antmicro/yosys/commit/ca8e0fd824d01a9b8eef937458693423a968aad5
but this requires fix to inconsistency introduced in BIN field with: https://github.com/alainmarcel/Surelog/pull/940

After these 2 fixes, bitstream is working on hardware again.

PR #83 (2020-12-07): Remove ibex_fetch_fifo.sv from skip list
Fixes #78 
Fixes #60
closed 2020-12-07 by kamilrakoczy