chipsalliance/UHDM-integration-tests #113: The Ibex submodule is outdated

tgorochowik, 2020-12-21:
The Ibex submodule in this repository is 5 months old.

The last commit that still works appears to be: https://github.com/lowRISC/ibex/commit/8953d82ca42725f6fe9d8669a422995ee7d9e9dc

The following change breaks the build:
https://github.com/lowRISC/ibex/commit/1a9545baaf973c361d4ee5ac7b715d5957a4ceef#diff-c179907737569996eff5a53763dee97f323251b0bed22743345126b62f71d6ccR723

Errors from yosys:
```
1. Executing UHDM frontend.
ERROR: Encountered unhandled typespec in handle parameter: 638, node str: \MSTATUS_RST_VAL
make: *** [Makefile:13: lowrisc_ibex_top_artya7_surelog_0.1.blif] Error 1
```
It looks as though `localparams` are not properly handled here.

We need to:
* Fix the issues with the curret ibex versions
* Update the Ibex submodule
* Setup cyclic submodule updates (dependabot?) to find issues in new versions automatically

tgorochowik, 2020-12-21:
The "working" commit mentioned above ( lowRISC/ibex@8953d82 ) actually fails in Vivado at a later step:
https://github.com/lowRISC/ibex/commit/8953d82ca42725f6fe9d8669a422995ee7d9e9dc#diff-7290a764a1649d05221e96494db6c6dc7ec4612c47afaf81d23cafe7f047d8ddR1436

```
CRITICAL WARNING: [Project 1-486] Could not resolve non-primitive black box cell '$assert' instantiated as 'u_core/IllegalParamSecure' [(...)/surelog-uhdm/vivado/top_artya7.edif:112228]
CRITICAL WARNING: [Project 1-486] Could not resolve non-primitive black box cell '$initstate' instantiated as 'u_core/$initstate$1015' [(...)/surelog-uhdm/vivado/top_artya7.edif:112226]
```

Ref: https://github.com/lowRISC/ibex/commit/8953d82ca42725f6fe9d8669a422995ee7d9e9dc#diff-7290a764a1649d05221e96494db6c6dc7ec4612c47afaf81d23cafe7f047d8ddR1436

kgugala, 2020-12-21:
the `$assert` and `$initstate` doesn't seem to hit us with current ibex master.

This commit is needed with the latest Ibex code https://github.com/antmicro/yosys/commit/d6a7549fb5420e9b6b49003b13790e584f868ae8

I did a quick check and removed failing `localparams` from Ibex' code. I was able to synthesize such design and get it working on FPGA, so this seems to be the only missing piece here

kbieganski, 2020-12-22:
I fixed these localparams with this commit: https://github.com/antmicro/yosys/commit/45096152246484ae8a709f404a6c7f5b52bc1bd2
but there's an issue in Surelog that needs to be resolved to support Ibex: https://github.com/alainmarcel/Surelog/issues/982
Working on it.

PR #114 (2020-12-28): Bump ibex
fixes #113 

@kbieganski once you solve https://github.com/alainmarcel/Surelog/issues/982 please update this PR.
closed 2020-12-28 by kgugala