chipsalliance/UHDM-integration-tests #156: ibex_simple_system: not working ram_2p.sv in verilator

kamilrakoczy, 2021-01-18:
Parsing ``ram_2p.sv`` (with ``prim_generic_ram_2p.sv`` and ``prim_ram_2p.sv``) file using UHDM frontend in verilator results in:
```
%Error-UNSUPPORTED: Unsupported: Slice of non-constant bounds
                            : ... In instance ibex_simple_system.u_ram.u_ram..u_impl_generic
%Error: Internal Error: ../V3Width.cpp:1015: AstSelPlus should disappear after widthSel
                                : ... In instance ibex_simple_system.u_ram.u_ram..u_impl_generic
                        ... See the manual and https://verilator.org for more assistance.
```

wtatarski, 2021-02-22:
Above error is fixed. However `ram_2p.sv` still doesn't work with verilator. For target `uhdm/verilator/test-ast` I got the following error:
```
%Error: Duplicate declaration of task: 'ram_2p.u_ram.u_impl_generic.'
        ... Location of original declaration
-node: TASK 0x561682dbba80 <e3450> {e0aa} u1=0x561682d15980  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
- V3Ast.cpp:1120:     Dumping obj_dir/Vram_2p_043_linkdot-preerr.tree
- V3LinkDot.cpp:973:     FUNC 0x561682dbbba0 <e29965> {e0aa} @dt=0x561682e24b60@(G/sw32)  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__simutil_set_mem
- V3LinkDot.cpp:973:     TASK 0x561682dbcc60 <e3561> {e0aa}  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
- V3LinkDot.cpp:269:  name ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
- V3LinkDot.cpp:270:  Var1 TASK 0x561682dbcc60 <e3561> {e0aa} u1=0x561682d15c00  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
- V3LinkDot.cpp:271:  Var2 TASK 0x561682dbba80 <e3450> {e0aa} u1=0x561682d15980  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
%Error: Duplicate declaration of task: 'ram_2p.u_ram.u_impl_generic.'
        ... Location of original declaration
-node: TASK 0x561682dbcc60 <e3561> {e0aa} u1=0x561682d15c00  ram_2p__DOT__u_ram__DOT__u_impl_generic__DOT__
```

At first glance I suspect that Surelog has some problem with parsing. Here is a part of UHDM which contains `@@BAD_SYMBOL@@`:
```
  |vpiTaskFunc:                                                                 
  \_function: (work@prim_generic_ram_2p.@@BAD_SYMBOL@@), line:21, parent:work@prim_generic_ram_2p
    |vpiAccessType:3                                                            
    |vpiVisibility:1                                                            
    |vpiDPICStr:2                                                               
    |vpiName:@@BAD_SYMBOL@@                                                     
    |vpiFullName:work@prim_generic_ram_2p.@@BAD_SYMBOL@@
```

wtatarski, 2021-03-12:
The issue with BAD_SYMBOL was solved with those 2 fixes in Surelog:
https://github.com/alainmarcel/Surelog/pull/1094
https://github.com/alainmarcel/Surelog/pull/1090

Also ram_2p module was added to build flow with Surelog

closed 2021-03-12 by wtatarski