chipsalliance/UHDM-integration-tests #205: Earlgrey: check status of every file for simulation via Surelog/UHDM -> Verilator

wtatarski, 2021-02-23:
The purpose of this issue is to track progress in successful simulation of Earlgrey in Verilator using Surelog.
List of Earlgrey files to check:
- [ ] top_pkg.sv
- [ ] gpiodpi.sv
- [ ] spidpi.sv
- [ ] uartdpi.sv
- [ ] usbdpi.sv
- [ ] dv_test_status_pkg.sv
- [ ] pins_if.sv
- [ ] ibex_pkg.sv
- [ ] entropy_src_pkg.sv
- [ ] jtag_pkg.sv
- [ ] rv_core_ibex_pkg.sv
- [ ] usbdev_pkg.sv
- [ ] prim_pkg.sv
- [ ] prim_secded_22_16_dec.sv
- [ ] prim_secded_22_16_enc.sv
- [ ] prim_secded_28_22_dec.sv
- [ ] prim_secded_28_22_enc.sv
- [ ] prim_secded_39_32_dec.sv
- [ ] prim_secded_39_32_enc.sv
- [ ] prim_secded_72_64_dec.sv
- [ ] prim_secded_72_64_enc.sv
- [ ] prim_secded_hamming_22_16_dec.sv
- [ ] prim_secded_hamming_22_16_enc.sv
- [ ] prim_secded_hamming_39_32_dec.sv
- [ ] prim_secded_hamming_39_32_enc.sv
- [ ] prim_secded_hamming_72_64_dec.sv
- [ ] prim_secded_hamming_72_64_enc.sv
- [ ] prim_util_pkg.sv
- [ ] top_earlgrey_pkg.sv
- [ ] dmidpi.sv
- [ ] jtagdpi.sv
- [ ] sw_test_status_pkg.sv
- [ ] sw_test_status_if.sv
- [ ] ibex_tracer_pkg.sv
- [ ] ibex_tracer.sv
- [ ] lc_ctrl_state_pkg.sv
- [ ] otbn_pkg.sv
- [ ] prim_diff_decode.sv
- [ ] prim_gf_mult.sv
- [ ] tlul_pkg.sv
- [ ] lc_ctrl_reg_pkg.sv
- [ ] lc_ctrl_pkg.sv
- [ ] otbn_tracer.sv
- [ ] otbn_trace_if.sv
- [ ] prim_generic_buf.sv
- [ ] prim_generic_clock_buf.sv
- [ ] prim_generic_clock_gating.sv
- [ ] prim_generic_clock_mux2.sv
- [ ] prim_generic_flop.sv
- [ ] prim_generic_flop_2sync.sv
- [ ] prim_generic_pad_wrapper.sv
- [ ] prim_generic_ram_1p.sv
- [ ] prim_generic_ram_2p.sv
- [ ] prim_generic_rom.sv
- [ ] prim_lfsr.sv
- [ ] prim_otp_pkg.sv
- [ ] prim_dom_and_2share.sv
- [ ] prim_xilinx_buf.sv
- [ ] prim_xilinx_clock_buf.sv
- [ ] prim_xilinx_clock_gating.sv
- [ ] prim_xilinx_clock_mux2.sv
- [ ] prim_xilinx_flop.sv
- [ ] prim_xilinx_pad_wrapper.sv
- [ ] alert_handler_reg_pkg.sv
- [ ] alert_handler_reg_top.sv
- [ ] pwrmgr_reg_pkg.sv
- [ ] pwrmgr_reg_top.sv
- [ ] rstmgr_reg_pkg.sv
[... 333 lines cut ...]
- [ ] rv_plic_reg_top.sv
- [ ] rv_plic.sv
- [ ] sensor_ctrl.sv
- [ ] flash_ctrl_erase.sv
- [ ] flash_ctrl_prog.sv
- [ ] flash_ctrl_rd.sv
- [ ] flash_ctrl_arb.sv
- [ ] flash_ctrl_info_cfg.sv
- [ ] flash_ctrl_lcmgr.sv
- [ ] flash_mp.sv
- [ ] flash_mp_data_region_sel.sv
- [ ] flash_phy.sv
- [ ] flash_phy_core.sv
- [ ] flash_phy_rd.sv
- [ ] flash_phy_prog.sv
- [ ] flash_phy_erase.sv
- [ ] flash_phy_rd_buffers.sv
- [ ] flash_phy_scramble.sv
- [ ] rstmgr.sv
- [ ] rstmgr_ctrl.sv
- [ ] rstmgr_por.sv
- [ ] rstmgr_crash_info.sv
- [ ] top_earlgrey_rnd_cnst_pkg.sv
- [ ] top_earlgrey.sv
- [ ] top_earlgrey_verilator.sv

RRozak, 2021-07-12:
Continuation in https://github.com/chipsalliance/UHDM-integration-tests/issues/363

RRozak, 2021-09-27:
Removed all modules from the list in https://github.com/antmicro/verilator/pull/496

closed 2021-09-27 by RRozak