chipsalliance/UHDM-integration-tests #233: Earlgrey: ibex_cs_registers.sv doesn't work in verilator

rkapuscik, 2021-03-11:
Parsing `ibex_cs_registers.sv` file results in error in Verilator:
```
%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:344: Null item passed to setOp1p
```
This happens because `ibex_counter.sv` is not yet parsed by Surelog (see https://github.com/alainmarcel/uhdm-integration/issues/208). but can also mask other errors.

kbieganski, 2021-03-12:
Investigating this together with https://github.com/alainmarcel/uhdm-integration/issues/208, as enabling both modules causes both errors to disappear, and a new error occurs:
```
%Error-UNSUPPORTED: Unsupported/Illegal: Assignment pattern member not underneath a supported construct: REPLICATE
                            : ... In instance top_earlgrey_verilator.top_earlgrey.u_rv_core_ibex.u_core.cs_registers_i
```

rkapuscik, 2021-03-12:
This error is a known issue due to the way Surelog handles parameters and was described here: https://github.com/alainmarcel/Surelog/issues/1013. It probably can be fixed by passing `--disable-feature=parametersubstitution` flag to Surelog.

kbieganski, 2021-03-15:
Fixed by https://github.com/antmicro/verilator/pull/90/

closed 2021-03-15 by kbieganski