chipsalliance/UHDM-integration-tests #234: Earlgrey: ibex_core.sv doesn't work in Verilator

rkapuscik, 2021-03-11:
Parsing `ibex_core.sv` file results in multiple warnings and errors in Verilator:
```
%Warning-DECLFILENAME: Filename 'ibex_alu' does not match MODULE name: 'ibex_alu_alu_i8'                                                                                                                                                                               [63/1820]
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-DECLFILENAME: Filename 'ibex_compressed_decoder' does not match MODULE name: 'ibex_compressed_decoder_compressed_decoder_i1'
%Warning-DECLFILENAME: Filename 'ibex_controller' does not match MODULE name: 'ibex_controller_controller_i6'                                    
%Warning-DECLFILENAME: Filename 'ibex_decoder' does not match MODULE name: 'ibex_decoder_decoder_i5'
%Warning-DECLFILENAME: Filename 'ibex_ex_block' does not match MODULE name: 'ibex_ex_block_ex_block_i7'
%Warning-DECLFILENAME: Filename 'ibex_fetch_fifo' does not match MODULE name: 'ibex_fetch_fifo_fifo_i3'                                 
%Warning-DECLFILENAME: Filename 'ibex_id_stage' does not match MODULE name: 'ibex_id_stage_id_stage_i4'
%Warning-DECLFILENAME: Filename 'ibex_if_stage' does not match MODULE name: 'ibex_if_stage_if_stage_i0'
%Warning-DECLFILENAME: Filename 'ibex_load_store_unit' does not match MODULE name: 'ibex_load_store_unit_load_store_unit_i10'        
%Warning-DECLFILENAME: Filename 'ibex_multdiv_fast' does not match MODULE name: 'ibex_multdiv_fast_multdiv_i9'
%Warning-DECLFILENAME: Filename 'ibex_prefetch_buffer' does not match MODULE name: 'ibex_prefetch_buffer_prefetch_buffer_i2'
%Warning-DECLFILENAME: Filename 'ibex_register_file_ff' does not match MODULE name: 'ibex_register_file_ff_register_file_i12'
%Warning-DECLFILENAME: Filename 'ibex_wb_stage' does not match MODULE name: 'ibex_wb_stage_wb_stage_i11'
%Warning-PINMISSING: ../src/lowrisc_dv_sim_sram_0/sim_sram.sv:100:7: Cell has missing pin: 'en_ifetch_i'
  100 |     ) u_tl_adapter_sim_sram (                                                                                                       
      |       ^~~~~~~~~~~~~~~~~~~~~                                                                                 
%Warning-DECLFILENAME: ../src/pulp-platform_riscv-dbg_0.1_0/pulp_riscv_dbg/src/dm_pkg.sv:19:9: Filename 'dm_pkg' does not match PACKAGE name: 'dm'
   19 | package dm;                                                                                  
      |         ^~                                                                               
%Warning-PINMISSING: ../src/lowrisc_systems_top_earlgrey_verilator_0.1/rtl/top_earlgrey_verilator.sv:103:16: Cell has missing pin: 'clk_main_jitter_en_o'
  103 |   top_earlgrey top_earlgrey (                                                                
      |                ^~~~~~~~~~~~                                                              
%Warning-VARHIDDEN: ../src/lowrisc_ip_rstmgr_0.1/rtl/rstmgr_ctrl.sv:11:17: Declaration of signal hides declaration in upper scope: 'PowerDomains'
   11 |   parameter int PowerDomains = 2,
      |                 ^~~~~~~~~~~~                                                            
                    ../src/lowrisc_systems_rstmgr_pkg_0.1/rtl/autogen/rstmgr_pkg.sv:18:17: ... Location of original declaration
   18 |   parameter int PowerDomains = 2;                                                                                                              
      |                 ^~~~~~~~~~~~                                                                                                                   
%Warning-VARHIDDEN: ../src/lowrisc_ip_rstmgr_0.1/rtl/rstmgr_ctrl.sv:12:18: Declaration of signal hides declaration in upper scope: 'OffDomains'
   12 |   localparam int OffDomains = PowerDomains - 1
      |                  ^~~~~~~~~~                                                             
                    ../src/lowrisc_systems_rstmgr_pkg_0.1/rtl/autogen/rstmgr_pkg.sv:23:17: ... Location of original declaration
   23 |   parameter int OffDomains = PowerDomains-1;
      |                 ^~~~~~~~~~
%Warning-VARHIDDEN: ../src/lowrisc_ip_flash_ctrl_0.1/rtl/flash_mp.sv:60:18: Declaration of signal hides declaration in upper scope: 'HwInfoRules'
   60 |   localparam int HwInfoRules = 3;
      |                  ^~~~~~~~~~~
                    ../src/lowrisc_systems_flash_ctrl_pkg_0.1/rtl/autogen/flash_ctrl_pkg.sv:173:17: ... Location of original declaration
  173 |   parameter int HwInfoRules = 3;
      |                 ^~~~~~~~~~~
%Warning-VARHIDDEN: ../src/lowrisc_ip_csrng_0.1/rtl/csrng.sv:15:17: Declaration of signal hides declaration in upper scope: 'NHwApps'
   15 |   parameter int NHwApps = 2
      |                 ^~~~~~~
                    ../src/lowrisc_ip_csrng_pkg_0.1/rtl/csrng_reg_pkg.sv:10:17: ... Location of original declaration
   10 |   parameter int NHwApps = 2;
      |                 ^~~~~~~
%Warning-IMPLICIT: ../src/lowrisc_ip_pattgen_0.1/rtl/pattgen_chan.sv:110:10: Signal definition not found, creating implicitly: 'complete_en'
                                                                           : ... Suggested alternative: 'complete_d'
  110 |   assign complete_en = (rep_cnt_en & (rep_cnt_q == reps_q)) | (~enable);
      |          ^~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:221:6: Pin not found: 'rvfi_valid'
  221 |     .rvfi_valid,
      |      ^~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:222:6: Pin not found: 'rvfi_order'
  222 |     .rvfi_order,
      |      ^~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:223:6: Pin not found: 'rvfi_insn'
  223 |     .rvfi_insn,
      |      ^~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:224:6: Pin not found: 'rvfi_trap'
  224 |     .rvfi_trap,
      |      ^~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:225:6: Pin not found: 'rvfi_halt'
  225 |     .rvfi_halt,
      |      ^~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:226:6: Pin not found: 'rvfi_intr'
  226 |     .rvfi_intr,
[... 29 lines cut ...]
  236 |     .rvfi_rd_wdata,
      |      ^~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:237:6: Pin not found: 'rvfi_pc_rdata'
  237 |     .rvfi_pc_rdata,
      |      ^~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:238:6: Pin not found: 'rvfi_pc_wdata'
  238 |     .rvfi_pc_wdata,
      |      ^~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:239:6: Pin not found: 'rvfi_mem_addr'
  239 |     .rvfi_mem_addr,
      |      ^~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:240:6: Pin not found: 'rvfi_mem_rmask'
  240 |     .rvfi_mem_rmask,
      |      ^~~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:241:6: Pin not found: 'rvfi_mem_wmask'
  241 |     .rvfi_mem_wmask,
      |      ^~~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:242:6: Pin not found: 'rvfi_mem_rdata'
  242 |     .rvfi_mem_rdata,
      |      ^~~~~~~~~~~~~~
%Error: ../src/lowrisc_ip_rv_core_ibex_0.1/rtl/rv_core_ibex.sv:243:6: Pin not found: 'rvfi_mem_wdata'
  243 |     .rvfi_mem_wdata,
      |      ^~~~~~~~~~~~~~
```
This file also depends on other Ibex modules, so it should be revisited after fixing #208 and #233.

rkapuscik, 2021-03-22:
Needs #209.

RRozak, 2021-09-10:
Removed from the lists in https://github.com/antmicro/verilator/pull/469/

closed 2021-09-10 by RRozak