chipsalliance/UHDM-integration-tests #240: Earlgrey: prim_flop.sv doesn't work in Verilator

rkapuscik, 2021-03-12:
Parsing prim_flop.sv file fails in Verilator:
```
%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:344: Null item passed to setOp1p
```
This file requires prim_generic_flop.sv (see https://github.com/alainmarcel/uhdm-integration/issues/238) and prim_xilinx_flop.sv to be parsed together, it should be revisited then.

rkapuscik, 2021-03-22:
Individual test passes (https://github.com/antmicro/verilator/pull/115), but needs to be parsed together with following files:
* lowrisc_ip_aes_0.6/rtl/aes_cipher_control.sv
* lowrisc_ip_csrng_0.1/rtl/csrng_ctr_drbg_gen.sv
* lowrisc_ip_csrng_0.1/rtl/csrng_cmd_stage.sv
* lowrisc_ip_csrng_0.1/rtl/csrng_main_sm.sv
* lowrisc_ip_entropy_src_0.1/rtl/entropy_src_main_sm.sv
* lowrisc_ip_entropy_src_0.1/rtl/entropy_src_ack_sm.sv
* lowrisc_ip_edn_0.1/rtl/edn_main_sm.sv
* lowrisc_ip_edn_0.1/rtl/edn_ack_sm.sv
* lowrisc_ip_aes_0.6/rtl/aes_control.sv
* lowrisc_ip_aes_0.6/rtl/aes_ctr.sv
* lowrisc_prim_generic_otp_0/rtl/prim_generic_otp.sv
* lowrisc_ip_rstmgr_0.1/rtl/rstmgr_por.sv
* lowrisc_ip_rstmgr_0.1/rtl/rstmgr_ctrl.sv
* lowrisc_ip_lc_ctrl_0.1/rtl/lc_ctrl_fsm.sv
* lowrisc_ip_pwrmgr_0.1/rtl/pwrmgr_fsm.sv

RRozak, 2021-09-27:
Removed from the list in https://github.com/antmicro/verilator/pull/496

closed 2021-09-27 by RRozak