chipsalliance/UHDM-integration-tests #249: Earlgrey: prim_flop_2sync.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_flop_2sync.sv file fails in Verilator:

%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:344: Null item passed to setOp1p

This file requires prim_generic_flop_2sync.sv (see #241) to be parsed, it should be revisited then.

rkapuscik, 2021-03-22:
Individual test passes (https://github.com/antmicro/verilator/pull/115), but needs to be parsed together with following files:
* lowrisc_prim_all_0.1/rtl/prim_sync_reqack.sv
* lowrisc_prim_all_0.1/rtl/prim_fifo_async.sv
* lowrisc_prim_lc_sync_0.1/rtl/prim_lc_sync.sv
* lowrisc_prim_all_0.1/rtl/prim_pulse_sync.sv
* lowrisc_ip_rstmgr_0.1/rtl/rstmgr_por.sv
* lowrisc_ip_rstmgr_0.1/rtl/rstmgr_ctrl.sv
* lowrisc_ip_flash_ctrl_0.1/rtl/flash_ctrl_lcmgr.sv
* lowrisc_ip_pinmux_component_0.1/rtl/pinmux_wkup.sv
* lowrisc_ip_usbdev_0.1/rtl/usbdev_aon_wake.sv
* lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv
* lowrisc_ip_usbdev_0.1/rtl/usbdev_flop_2syncpulse.sv
* lowrisc_ip_uart_0.1/rtl/uart_core.sv
* lowrisc_ip_pwrmgr_0.1/rtl/pwrmgr_cdc.sv

RRozak, 2021-09-27:
Removed from the list in https://github.com/antmicro/verilator/pull/496

closed 2021-09-27 by RRozak