chipsalliance/UHDM-integration-tests #251: Earlgrey: prim_clock_mux2.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_clock_mux2.sv file through Surelog-UHDM results in errors in Verilator:
```
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:129:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:137:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:145:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:198:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:206:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: ../src/lowrisc_ip_usbdev_0.1/rtl/usbdev_iomux.sv:214:3: Cannot find file containing module: 'prim_generic_clock_mux2'
%Error: Exiting due to 6 error(s)
```
even though prim_generic_clock_mux2.sv is parsed without errors.

rkapuscik, 2021-03-30:
Added individual module test (#283) that passes in Verilator (https://github.com/antmicro/verilator/pull/129), but the module needs to be enabled along with usbdev_iomux.sv

RRozak, 2021-09-14:
Removed from the lists in https://github.com/antmicro/verilator/pull/475

closed 2021-09-14 by RRozak