chipsalliance/UHDM-integration-tests #252: Earlgrey: prim_clock_gating_sync.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_clock_gating_sync.sv file through Surelog-UHDM results in errors in Verilator:
```
%Error: Cannot find file containing module: 'i_sync'
%Error: Cannot find file containing module: 'i_cg'
%Error: Exiting due to 2 error(s)
```
This file requires prim_flop_2sync.sv (see #249) and prim_clock_gating (see #209) to be parsed, it should be revisited then.

RRozak, 2021-04-07:
I created test for that module: https://github.com/antmicro/uhdm-integration/commit/a275e8cdc0e8f8f88ccea1b5de629b7d446b9574

RRozak, 2021-04-08:
Test passed: https://github.com/antmicro/verilator/actions/runs/729519765

RRozak, 2021-04-12:
The module `prim_flop` should be enabled along with files:
- entropy_src_main_sm.sv
- aes_ctr.sv
- csrng_cmd_stage.sv
- edn_main_sm.sv
- otp_ctrl_scrmbl.sv
- otp_ctrl_dai.sv
- aes_cipher_control.sv
- prim_clock_div.sv
- aes_control.sv
- otp_ctrl_part_buf.sv
- otp_ctrl_lci.sv
- prim_generic_otp.sv
- pwrmgr_fsm.sv
- csrng_ctr_drbg_gen.sv
- csrng_main_sm.sv
- lc_ctrl_fsm.sv
- otp_ctrl_lfsr_timer.sv
- rstmgr_ctrl.sv
- edn_ack_sm.sv
- otp_ctrl_kdi.sv
- entropy_src_ack_sm.sv
- rstmgr_por.sv
- otp_ctrl_part_unbuf.sv
- flash_ctrl.sv
- csrng_ctr_drbg_upd.sv

The module `prim_generic_flop` should be enabled along with files:
- prim_lc_sender.sv

The module `prim_clock_gating` should be enabled along with files:
- clkmgr.sv
- ibex_core.sv

RRozak, 2021-08-19:
It was removed from Verilator list in https://github.com/antmicro/verilator/pull/432

closed 2021-08-19 by RRozak