chipsalliance/UHDM-integration-tests #253: Earlgrey: prim_alert_receiver.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_alert_receiver.sv file fails in Verilator:
```
%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:344: Null item passed to setOp1p
```
This file requires prim_flop_2sync.sv (see #249) to be parsed (through prim_diff_decode.sv), it should be revisited then.

RRozak, 2021-03-30:
I am currently working on it.

RRozak, 2021-03-30:
The `prim_buf` module needs to be enabled along with following files:

- aes_sel_buf_chk.sv
- prim_lc_sync.sv
- otp_ctrl_part_buf.sv
- otp_ctrl_part_unbuf.sv

RRozak, 2021-03-30:
Individual tests passes: https://github.com/antmicro/verilator/pull/142

RRozak, 2021-08-19:
Removed from Verilator list in https://github.com/antmicro/verilator/pull/433

closed 2021-08-19 by RRozak