chipsalliance/UHDM-integration-tests #261: Earlgrey: prim_esc_receiver.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_esc_receiver.sv file fails in Verilator:
```
%Error: ../src/lowrisc_ip_aes_0.6/rtl/aes_sel_buf_chk.sv:38:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_lc_sync_0.1/rtl/prim_lc_sync.sv:50:7: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_ip_otp_ctrl_0.1/rtl/otp_ctrl_part_buf.sv:625:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_ip_otp_ctrl_0.1/rtl/otp_ctrl_part_unbuf.sv:348:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:48:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:78:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:82:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:74:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:90:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:250:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:254:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:61:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:93:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:97:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:101:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:105:3: Cannot find file containing module: 'prim_buf'
%Error: Exiting due to 16 error(s)
```
This file requires prim_flop_2sync.sv (see #249) to be parsed (through prim_diff_decode.sv), it should be revisited then.

RRozak, 2021-03-29:
Individual tests pass: https://github.com/antmicro/verilator/pull/142

RRozak, 2021-08-19:
Removed from Verilator list in https://github.com/antmicro/verilator/pull/433

closed 2021-08-19 by RRozak