chipsalliance/UHDM-integration-tests #262: Earlgrey: prim_esc_sender.sv doesn't work in Verilator

rkapuscik, 2021-03-17:
Parsing prim_esc_sender.sv file fails in Verilator:
```
%Error: ../src/lowrisc_ip_aes_0.6/rtl/aes_sel_buf_chk.sv:38:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_lc_sync_0.1/rtl/prim_lc_sync.sv:50:7: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_ip_otp_ctrl_0.1/rtl/otp_ctrl_part_buf.sv:625:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_ip_otp_ctrl_0.1/rtl/otp_ctrl_part_unbuf.sv:348:5: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:48:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:78:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv:82:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:74:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:90:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:250:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv:254:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:61:3: Cannot find file containing module: 'prim_diff_decode'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:93:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:97:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:101:3: Cannot find file containing module: 'prim_buf'
%Error: ../src/lowrisc_prim_all_0.1/rtl/prim_alert_receiver.sv:105:3: Cannot find file containing module: 'prim_buf'
%Error: Exiting due to 16 error(s)
```
This file requires prim_flop_2sync.sv (see #249) to be parsed (through prim_diff_decode.sv), it should be revisited then.

RRozak, 2021-03-29:
Individual test passes (https://github.com/antmicro/verilator/pull/142), but needs to be parsed together with following files:

- lowrisc_prim_abstract_prim_pkg_0.1/prim_pkg.sv
- lowrisc_prim_all_0.1/rtl/prim_esc_pkg.sv
- lowrisc_prim_all_0.1/rtl/prim_esc_sender.sv 
- lowrisc_prim_diff_decode_0/rtl/prim_diff_decode.sv
- lowrisc_prim_abstract_flop_2sync_0/prim_flop_2sync.sv
- lowrisc_prim_generic_flop_2sync_0/rtl/prim_generic_flop_2sync.sv
- lowrisc_prim_abstract_flop_0/prim_flop.sv
- lowrisc_prim_generic_buf_0/rtl/prim_generic_buf.sv
- lowrisc_prim_abstract_buf_0/prim_buf.sv

RRozak, 2021-09-03:
Already removed from the lists.

closed 2021-09-03 by RRozak