chipsalliance/UHDM-integration-tests #265: Investigate segmentation faults of UhdmYosys in sv-tests

hzeller, 2021-03-17:
As discussed offline, there might be some inconsistencies of a manually built Yosys+UHDM vs. the Conda builds used in 
https://github.com/SymbiFlow/sv-tests CIs.

There are some of these tests that plain segfault in sv-tests, currently, such as Ibex
https://symbiflow.github.io/sv-tests-results/#UhdmYosys|ibex|ibex

Segmentation faults usually come out as exit code 139, so here is how to get a more complete list
```bash
wget -q -Oreport.csv https://symbiflow.github.io/sv-tests-results/report.csv
grep UhdmYosys,False,139, report.csv
```

kgugala, 2021-03-17:
see https://github.com/hdl/conda-eda/issues/96

kamilrakoczy, 2021-03-24:
The reason why Ibex test exited with segfault was wrong used target in fusesoc to generate file list. I've created WIP PR that fixes this issue: https://github.com/SymbiFlow/sv-tests/pull/1412

kamilrakoczy, 2021-03-31:
Above PR is merged and Ibex is now green in sv-tests, but there are still some segmentation faults.
I've created PR to yosys that is fixing wire declaration inside initial block. This fixes several of them (especially enum tests): https://github.com/antmicro/yosys/pull/194

kamilrakoczy, 2021-05-07:
After: https://github.com/SymbiFlow/sv-tests/pull/1502 only 27 segmentation faults left:
```
hdlconvertor_std2017_p137_2,UhdmYosys,False,139,hdlconv_std2017,751,30,0.241055,0.01007,0.251125,24.156
hdlconvertor_std2017_p327,UhdmYosys,False,139,hdlconv_std2017,859,30,0.542764,0.086029,0.628793,142.934
regress-vlg_function5_iv,UhdmYosys,False,139,ivtest,768,30,0.291289,0.019719,0.311008,32.305
regress-vlg_event2_iv,UhdmYosys,False,139,ivtest,1049,30,0.136821,0.012719,0.14954,22.84
regress-sv_br_gh104b_iv,UhdmYosys,False,139,ivtest,342,30,0.622613,0.017443,0.640056,34.383
regress-sv_br_gh104a_iv,UhdmYosys,False,139,ivtest,396,30,0.690188,0.010424,0.700612,34.52
regress-sv_br_gh418_iv,UhdmYosys,False,139,ivtest,328,30,0.434462,0.07165,0.506112,142.812
regress-vlg_br_gh345_iv,UhdmYosys,False,139,ivtest,3000,30,2.018912,0.024951,2.043863,52.156
regress-vlg_contrib82_iv,UhdmYosys,False,139,ivtest,1524,30,0.569661,0.015856,0.585517,32.102
swerv,UhdmYosys,False,139,swerv,1263463,180,38.506946,0.771699,39.278645,986.996
BaseJumpSTL_bsg_tag_trace_replay,UhdmYosys,False,139,basejump,28750,30,1.810091,0.038819,1.84891,55.535
BaseJumpSTL_bsg_wormhole_router_adapter_in,UhdmYosys,False,139,basejump,25709,30,1.387303,0.047902,1.435205,56.922
BaseJumpSTL_bsg_channel_tunnel_out,UhdmYosys,False,139,basejump,27275,30,1.494052,0.040515,1.534567,61.445
BaseJumpSTL_bsg_wormhole_router_adapter_out,UhdmYosys,False,139,basejump,26178,30,1.35394,0.038203,1.392143,59.355
BaseJumpSTL_bsg_nonsynth_dpi_rom,UhdmYosys,False,139,basejump,27670,30,1.483906,0.047718,1.531624,49.359
BaseJumpSTL_bsg_comm_link,UhdmYosys,False,139,basejump,57676,30,3.577377,0.043892,3.621269,85.867
hdlconvertor_std2012_p102,UhdmYosys,False,139,hdlconv_std2012,791,30,0.251862,0.003214,0.255076,24.016
bp_fe,UhdmYosys,False,139,black-parrot,1380466,300,81.014689,3.608257,84.622946,3888.703
asicworld_code_hdl_models_parity_using_function,UhdmYosys,False,139,yosys,1166,30,0.305991,0.026664,0.332655,32.008
simple_const_func_shadow,UhdmYosys,False,139,yosys,979,30,0.386325,0.012007,0.398332,27.297
simple_func_recurse,UhdmYosys,False,139,yosys,798,30,0.312069,0.076087,0.388156,142.555
function_recursive,UhdmYosys,False,139,13.4.2,1012,30,0.651741,0.108372,0.760113,142.457
jump_return,UhdmYosys,False,139,12.8 uvm-req,540,30,0.41987,0.02474,0.44461,33.191
structure-replication,UhdmYosys,False,139,5.10,535,30,0.192661,0.004932,0.197593,24.137
arrays-key-index,UhdmYosys,False,139,5.11,397,30,0.089161,0.001436,0.090597,20.332
named_event_wait,UhdmYosys,False,139,15.5 uvm-req,514,30,0.105671,0.010329,0.116,21.57
class_test_27,UhdmYosys,False,139,6.15 8.3 uvm-req,456,30,0.044708,0.012847,0.057555,19.035
```

I think, now it is related to some unsupported feature in yosys/Surelog. I will check and try to fix them.