chipsalliance/UHDM-integration-tests #321: Makefile issues

udif, 2021-05-07:
I'm trying to debug an issue in (https://github.com/SymbiFlow/sv-tests/issues/1501) which is triggered by sv-tests' runners.mk:
```
	cd $(RDIR)/uhdm-integration && $(MAKE) image/bin/surelog
```
Looking at the uhdh-integration Makefile I see no rule for `image/bin/surelog` which could be stteributed to an sv-tests bug, but I still see things I don't understand in the Makefile:
* All the XXX_BIN vars point to `${root_dir}/../image/bin/XXX` but the clean rule points to `$(root_dir)/image`, not `$(root_dir)/../image`
* Are all the XXX_BIN tool references point to plain builds, built externally by their respective tools, or are these modified builds in some way, as explained in `https://github.com/alainmarcel/uhdm-integration#readme` ?

rkapuscik, 2021-05-07:
Good catch, the `make clean` target is not valid here anymore. This is because the modified tools (Verilator and Yosys with added UHDM support) used to be submodules in this repository, but are now used on their own and this repository hosts only tests.
* #323 will take care of cleaning the makefile here
* tool references expect modified versions of [Verilator](https://github.com/antmicro/verilator) and [Yosys](https://github.com/antmicro/yosys/). Those tools will also build their own Surelog binary that is guaranteed to be compatible and store it in their `image` folder to avoid interfering with system-wide installed tools.

udif, 2021-05-07:
Thanks for your reply.
What is the recommended build flow for someone starting from scratch for a working Surelog and its related projects in particular, and the whole `sv-tests` repo in general? Obviously the current build instructions are outdated.

What is the roadmap for the Surelog integration in Verilator and Yosys? I see that the antmicro repos are not github forks and so it is harder to see as to what extent are they trying to keep it aligned with the original projects