chipsalliance/UHDM-integration-tests #340: Error when running "make uhdm/yosys/test-ast"

peijunh, 2021-06-13:
I follow the step of Verilator setup and Yosys setup. Then I run "make uhdm/yosys/test-ast" in uhdm-integration for both Yosys and Verilator. However, both of them don't work. 

The Error for Yosys is shown below:

rm -rf /home/houpeijun/peijunh/yosys/uhdm-integration/build
mkdir -p /home/houpeijun/peijunh/yosys/uhdm-integration/build
(cd /home/houpeijun/peijunh/yosys/uhdm-integration/build && \
        /home/houpeijun/peijunh/yosys/uhdm-integration/../image/bin/surelog -parse -sverilog -d coveruhdm   /home/houpeijun/peijunh/yosys/uhdm-integration/tests/onenet/top.sv)
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[FTL:CM0008] Verilog File "/usr/local/lib/surelog/sv/builtin.sv" does not exist.

[  FATAL] : 1
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
make: *** [surelog/parse] Error 1
*********************************************************************************
*********************************************************************************
*********************************************************************************
And the Error for Verilator is shown below:

make uhdm/verilator/test-ast
rm -rf /home/houpeijun/peijunh/verilator/uhdm-integration/build
mkdir -p /home/houpeijun/peijunh/verilator/uhdm-integration/build
(cd /home/houpeijun/peijunh/verilator/uhdm-integration/build && \
        /home/houpeijun/peijunh/verilator/uhdm-integration/../image/bin/surelog -parse -sverilog -d coveruhdm   /home/houpeijun/peijunh/verilator/uhdm-integration/tests/onenet/top.sv)
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[FTL:CM0008] Verilog File "/usr/local/lib/surelog/sv/builtin.sv" does not exist.

[  FATAL] : 1
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
make: *** [surelog/parse] Error 1

alaindargelas, 2021-06-16:
Your Surelog install directory is not following the expected location (/usr/local/bin/surelog) after intall.
Please add -nobuiltin to the Surelog command line in the meantime.
Check your install path.

msaligane, 2021-06-23:
Merci @alaindargelas. This is resolved.

closed 2021-06-23 by alaindargelas