chipsalliance/UHDM-integration-tests #341: Compiling OpenTitan with UHDM

zyedidia, 2021-06-13:
I'm not sure if this is the right place for this issue, so let me know if I should open this issue elsewhere. 

I'd like to compile the [OpenTitan](https://github.com/lowRISC/opentitan/tree/8b9fe4bf2db8ccfac0b26600decf07cf41867e07) core with SureLog+UHDM+Yosys, and I'm wondering what the status is. I have managed to generate a UHDM database file using SureLog, but when using antmicro/yosys for synthesis I get the following error:

```
1. Executing UHDM frontend.
Internal Error: adding wrong object type (logic_var) in a enum_struct_union_packed_var_group group!

Internal Error: adding wrong object type (logic_var) in a enum_struct_union_packed_var_group group!

VPI ERROR: Bad usage of vpi_get
```

Is UHDM support in Yosys not yet ready for this, or am I doing something incorrect? I essentially ran SureLog on all the OpenTitan source files, and ran the result through a yosys script which uses `read_uhdm` and tries to perform synthesis. Happy to provide more information if helpful. Thanks!

mithro, 2021-06-13:
There isn't such a thing as "OpenTitan core" - do you mean the Early Grey configuration of OpenTitan? If so there are still issues which need to be fixed which can be seen at https://github.com/alainmarcel/uhdm-integration/issues?q=is%3Aopen+is%3Aissue+label%3Ayosys

zyedidia, 2021-06-13:
Yes I'm using the Earlgrey configuration. Thanks for the link, I will keep an eye on those issues then!

closed 2021-06-13 by zyedidia