chipsalliance/UHDM-integration-tests #343: Equivalence checking of Ibex in Yosys

kbieganski, 2021-06-21:
We would like a way to automatically verify (in a CI) that our Yosys-UHDM frontend is working correctly for Ibex. Equivalence checking in Yosys seems to be a good way to do it.

There is some work on that started here: https://github.com/antmicro/yosys/tree/uhdm-yosys-equiv-check

The idea is to equivalence check Ibex parsed with Surelog against Ibex parsed with Yosys' built-in parser (after putting it through sv2v).

Unfortunately, most of the modules are failing so far: https://github.com/antmicro/yosys/actions/runs/956724171

alaindargelas, 2021-06-23:
Fix the Surelog project definitions for each module.
They seem all wrong. Missing includes and all

kbieganski, 2021-06-29:
> Fix the Surelog project definitions for each module.
> They seem all wrong. Missing includes and all

In some cases, yeah, but even modules that don't have missing includes are failing, like `ibex_pmp`.