chipsalliance/UHDM-integration-tests #349: Trouble building yosys with the configure script

erictaur, 2021-06-24:
I encountered several errors that look like these when building yosys with the configure script:

```
In file included from frontends/uhdm/UhdmAst.cc:6:

In file included from /shared/yosys/image/include/uhdm/headers/uhdm.h:69: 

In file included from /shared/yosys/image/include/uhdm/headers/vpi_uhdm.h:30:

In file included from /usr/bin/../lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/unordered_map:47:

In file included from /usr/bin/../lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/hashtable.h:35:

/usr/bin/../lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/hashtable_policy.h:830:23: 

error: implicit instantiation of undefined template 'std::hash<Yosys::AST::AstNodeType>'
bool __use_ebo = !__is_final(_Tp) && __is_empty(_Tp)>
```

Other tools such as surelog, uhdm-dump, uhdm-hier are built successfully and can be located under `images/bin`.

Version of tools used for the build:
`gcc (GCC) 7.3.1 20180303 (Red Hat 7.3.1-5)`
`cmake version 3.20.5`

All prerequisites required by Yosys are fulfilled (installed)

Thanks!

erictaur, 2021-06-28:
The distro I'm using is CentOS if that helps :)

kgugala, 2021-06-28:
@erictaur you log suggest you're using GCC 4.8.5 rather than 7.3.1. 4.8.5 is quite old and you may not be able to build Yosys with it.

erictaur, 2021-06-29:
When I do `which gcc` in the terminal, this is showing up: `gcc (GCC) 7.3.1 20180303 (Red Hat 7.3.1-5)`
So if the logs suggest otherwise then it is pretty weird since I specifically specified to use devtoolset-7 when building yosys
Is there a file that specifies the location of gcc it is using by default?