chipsalliance/UHDM-integration-tests #387: build_binaries.sh fails while creating UVM precompiled PKG

EngRaff92, 2021-08-01:
Consolidate compiler generated dependencies of target surelog-bin
[ 97%] Built target surelog-bin
[ 97%] Generating bin/pkg/work/uvm_pkg.sv.slpa
       Creating UVM precompiled package...
make[3]: *** [bin/pkg/work/uvm_pkg.sv.slpa] Error 3
make[3]: *** Deleting file `bin/pkg/work/uvm_pkg.sv.slpa'
make[2]: *** [CMakeFiles/PrecompileUVM.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
+ autoconf
autoconf: error: no input file
++ nproc
./build_binaries.sh: line 5: nproc: command not found
+ make -C vcddiff PREFIX=./image -j
make: *** vcddiff: No such file or directory.  Stop.

I'm not able to get the binaries working causing Verilator not fully installed.

Any clue on how I can get it sorted out ?
Regards

kgugala, 2021-08-01:
Hi @EngRaff92, can you attach a full log? Also, what OS are you using? From the short log you attached it looks like you don't have `nproc`. `nproc` is a part of `GNU coreutils` package and shoul be available on most Linux distributions

EngRaff92, 2021-08-01:
Here is the full log:
./build_binaries.sh
+ cd Surelog
++ sysctl -n hw.logicalcpu
+ make PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ release install -j 8
make: nproc: Command not found
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ -S . -B build
-- CMAKE_CXX_FLAGS:  -std=c++11 -stdlib=libc++ -Wall -pedantic -Werror -Wextra -Wno-unused-parameter
-- Proceeding with version: 1.12.0.445
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Building without demo. To enable demo build use: -DWITH_DEMO=True
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:31 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:32 (CMAKE_POLICY):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:33 (CMAKE_POLICY):
  The OLD behavior for policy CMP0042 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:38 (CMAKE_POLICY):
  The OLD behavior for policy CMP0059 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:39 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Output libraries to /Users/raffaelemac/verilator/Surelog/dist
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR Crypto SSL)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/raffaelemac/verilator/Surelog/build
cmake --build build -j 1
make[1]: warning: -jN forced in submake: disabling jobserver mode.
Consolidate compiler generated dependencies of target kj
[  5%] Built target kj
Consolidate compiler generated dependencies of target capnp
[  8%] Built target capnp
Consolidate compiler generated dependencies of target kj-async
[ 10%] Built target kj-async
Consolidate compiler generated dependencies of target capnp-json
[ 11%] Built target capnp-json
Consolidate compiler generated dependencies of target capnpc
[ 14%] Built target capnpc
Consolidate compiler generated dependencies of target capnp_tool
[ 14%] Built target capnp_tool
Consolidate compiler generated dependencies of target capnpc_cpp
[ 14%] Built target capnpc_cpp
[ 14%] Built target GenerateCode
Consolidate compiler generated dependencies of target uhdm
[ 25%] Built target uhdm
Consolidate compiler generated dependencies of target flatc
[ 32%] Built target flatc
[ 32%] Built target GenerateSerializer
[ 32%] Built target make_lib_output_dir
[ 33%] Built target utfcpp
Consolidate compiler generated dependencies of target antlr4_static
[ 67%] Built target antlr4_static
[ 67%] Built target GenerateParser
Consolidate compiler generated dependencies of target flatbuffers
[ 69%] Built target flatbuffers
Consolidate compiler generated dependencies of target surelog
[ 97%] Built target surelog
Consolidate compiler generated dependencies of target hellouhdm
[ 97%] Built target hellouhdm
Consolidate compiler generated dependencies of target surelog-bin
[ 97%] Built target surelog-bin
[ 97%] Generating bin/pkg/work/uvm_pkg.sv.slpa
       Creating UVM precompiled package...
make[3]: *** [bin/pkg/work/uvm_pkg.sv.slpa] Error 3
make[3]: *** Deleting file `bin/pkg/work/uvm_pkg.sv.slpa'
make[2]: *** [CMakeFiles/PrecompileUVM.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
+ autoconf
autoconf: error: no input file
++ sysctl -n hw.logicalcpu
+ make -C vcddiff PREFIX=./image -j 8
make: *** vcddiff: No such file or directory.  Stop.

I'm installing it on MAC M1, I replaced the nproc with sysctl -n hw.logicalcpu which is the equivalent to proc in MAC but it seems that even by using an alias or export I'm still not able to get it fully replaced. 

Regards

EngRaff92, 2021-08-01:
Seems fixed after doing ### brew install coreutils. Nevertheless still getting it not working fine:

╰─❯ ./build_binaries.sh
+ cd Surelog
++ sysctl -n hw.logicalcpu
+ make PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ release install -j 8
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ -S . -B build
-- CMAKE_CXX_FLAGS:  -std=c++11 -stdlib=libc++ -Wall -pedantic -Werror -Wextra -Wno-unused-parameter
-- Proceeding with version: 1.12.0.445
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Building without demo. To enable demo build use: -DWITH_DEMO=True
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:31 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:32 (CMAKE_POLICY):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:33 (CMAKE_POLICY):
  The OLD behavior for policy CMP0042 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:38 (CMAKE_POLICY):
  The OLD behavior for policy CMP0059 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:39 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Output libraries to /Users/raffaelemac/verilator/Surelog/dist
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR Crypto SSL)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/raffaelemac/verilator/Surelog/build
cmake --build build -j 8
make[1]: warning: -jN forced in submake: disabling jobserver mode.
Consolidate compiler generated dependencies of target flatbuffers
[... 31 lines cut ...]
[ 98%] Built target surelog
Consolidate compiler generated dependencies of target surelog-bin
Consolidate compiler generated dependencies of target hellodesign
Consolidate compiler generated dependencies of target hellouhdm
[100%] Built target hellouhdm
[100%] Built target surelog-bin
[100%] Built target hellodesign
[100%] Generating bin/pkg/work/ovm_pkg.sv.slpa
[100%] Generating bin/pkg/work/uvm_pkg.sv.slpa
       Creating UVM precompiled package...
       Creating OVM precompiled package...
       Package OVM created
[100%] Built target PrecompileOVM
make[3]: *** [bin/pkg/work/uvm_pkg.sv.slpa] Error 3
make[3]: *** Deleting file `bin/pkg/work/uvm_pkg.sv.slpa'
make[2]: *** [CMakeFiles/PrecompileUVM.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
+ autoconf
autoconf: error: no input file
++ sysctl -n hw.logicalcpu
+ make -C vcddiff PREFIX=./image -j 8
make: *** vcddiff: No such file or directory.  Stop.

Regards and thanks for the support

kgugala, 2021-08-01:
I see two issues:

1. Your CMake seems to be quite old - you can try updating it
1. you're missing [vcddiff](https://github.com/veripool/vcddiff)

EngRaff92, 2021-08-01:
I currently have the latest CMAKE version cmake version 3.21.1 and VCDDIFF is installed.

Two installation have been made of VCDDIFF:

1. one in my user dir and bin file has been updated with the binary
2. there is vcddiff dire in the Verilator dir and by doing make it install it.

Should I place the binary of vcddiff somewhere specifically so that make -C vcddiff works?

regards

EngRaff92, 2021-08-01:
Same issue after installing it in the local vcddiff dir and after placing it in the usr/local/bin dir. The issue is still there. Thanks for the support

EngRaff92, 2021-08-01:
╭─   ~/verilator   uhdm-verilator !6 ?2                                                         5s  15:59:36
╰─❯ ./build_binaries.sh
+ cd Surelog
++ sysctl -n hw.logicalcpu
+ make PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ release install -j 8
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/raffaelemac/verilator/Surelog/../image/ -S . -B build
-- CMAKE_CXX_FLAGS:  -std=c++11 -stdlib=libc++ -Wall -pedantic -Werror -Wextra -Wno-unused-parameter
-- Proceeding with version: 1.12.0.445
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Building without demo. To enable demo build use: -DWITH_DEMO=True
CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:31 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:32 (CMAKE_POLICY):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:33 (CMAKE_POLICY):
  The OLD behavior for policy CMP0042 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:38 (CMAKE_POLICY):
  The OLD behavior for policy CMP0059 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at third_party/antlr4/runtime/Cpp/CMakeLists.txt:39 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Output libraries to /Users/raffaelemac/verilator/Surelog/dist
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR Crypto SSL)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/raffaelemac/verilator/Surelog/build
cmake --build build -j 8
make[1]: warning: -jN forced in submake: disabling jobserver mode.
Consolidate compiler generated dependencies of target flatbuffers
[  1%] Built target utfcpp
[... 36 lines cut ...]
[100%] Built target hellodesign
[100%] Generating bin/pkg/work/uvm_pkg.sv.slpa
[100%] Generating bin/pkg/work/ovm_pkg.sv.slpa
       Creating UVM precompiled package...
       Creating OVM precompiled package...
       Package OVM created
[100%] Built target PrecompileOVM
make[3]: *** [bin/pkg/work/uvm_pkg.sv.slpa] Error 3
make[3]: *** Deleting file `bin/pkg/work/uvm_pkg.sv.slpa'
make[2]: *** [CMakeFiles/PrecompileUVM.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
+ autoconf
autoconf: error: no input file

Is the result after:
  1 #!/bin/bash
  2 set -ex
  3 cd Surelog && make PREFIX=$PWD/../image/ release install -j $(sysctl -n hw.logicalcpu) && cd ..
  4 autoconf && ./configure --prefix=$PWD/image && make -j $(sysctl -n hw.logicalcpu) && make install
  5 #make -C ./vcddiff PREFIX=./image -j $(sysctl -n hw.logicalcpu)
  6 #cp -p vcddiff/vcddiff image/bin/vcddiff
  7 #cp -p /usr/local/bin/vcddiff ./image/bin/vcddiff

I commented out the last 3 lines and copied the vcddiff manually in the image dir.

EngRaff92, 2021-08-01:
Lastly I successfully managed to the get it compiled and generated, seems like that if executed on the cmd line the pkg generation works (weird).

Anyway it seems like the Surelog bin cannot bi built fully on MACOS due to lack of supported LDFLAG like -lrt which is not available in MACOS. 

Since is just a matter of library (just lrt and apparently is not a huge hassle but please confirm it) could you please make it compatible with the library available on MACOS? or kind of provide a switch which changes the LDFLAGS once under MACOS. Thanks

tgorochowik, 2021-08-02:
Hi @EngRaff92 I am glad to hear it worked!

> Anyway it seems like the Surelog bin cannot bi built fully on MACOS due to lack of supported LDFLAG like -lrt which is not available in MACOS.
>
> Since is just a matter of library (just lrt and [...]

MacOS is not yet officially supported. Would you be willing to give it a go to see if modifying the flags is enough to compile and run Surelog? That would be greatly appreciated!

It would be nice if we could continue this discussion directly in a ticket in: https://github.com/chipsalliance/Surelog/blob/master/.github/workflows/main.yml

It your suggestion (modifying the flags) is indeed all it takes then the next step would be to add a CI job for macos to make sure it continues to build

EngRaff92, 2021-08-02:
Hello @tgorochowik yes the precompilation worked but it is still a bit tricky. Anyway, I'm not that good at this kind of stuff but I can give it a shot and help you guys since your work is amazing, and would be great to have it compiled on all the platforms.

tgorochowik, 2021-08-02:
Ahh, sorry for the wrong link, the link is to the CI setup where a mac os job would have to be added.

I'd like us to continue the discussion in an issue in the Surelog repo. This repo here is just for the tests and this problem regards Surelog itself to there it would reach more people that may be interested in supporting mac os

EngRaff92, 2021-08-02:
I get you now, yep i will open an issue in the repo and add a ref to this issue as well. Thanks so much