chipsalliance/UHDM-integration-tests #494: Earlgrey: newest prim_fifo_async doesn't work in Verilator

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, prim_fifo_async.sv is failing with:
```
%Error: /home/runner/work/verilator/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_fifo_0/rtl/prim_fifo_async.sv:255:35: Can't find definition of task/function: 'gray2dec'
%Error: /home/runner/work/verilator/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_fifo_0/rtl/prim_fifo_async.sv:257:35: Can't find definition of task/function: 'gray2dec'
%Error: /home/runner/work/verilator/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_fifo_0/rtl/prim_fifo_async.sv:259:31: Can't find definition of task/function: 'dec2gray'
%Error: /home/runner/work/verilator/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_fifo_0/rtl/prim_fifo_async.sv:260:31: Can't find definition of task/function: 'dec2gray'
%Error: Exiting due to 4 error(s)
```
See https://github.com/antmicro/verilator/pull/319/checks?check_run_id=3737443029.

Those functions are defined in that file.

RRozak, 2021-09-29:
The error can be reproduced by https://github.com/chipsalliance/UHDM-integration-tests/pull/517.
It is blocked by https://github.com/chipsalliance/Surelog/issues/1968

RRozak, 2021-10-01:
After Surelog upgrade there is an error: `%Error: Unknown object type: unsupported_typespec`

RRozak, 2021-10-01:
Blocked by https://github.com/chipsalliance/Surelog/issues/1986