chipsalliance/UHDM-integration-tests #497: Earlgrey: newest prim_flash doesn't work in Verilator

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, prim_flash.sv is failing with:
```
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_jtag_pkg_0.1/rtl/jtag_pkg.sv:8:11: '::' expected to reference a class/package but referenced '<unresolved-object>'

                : ... Suggest '.' instead of '::'
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_jtag_pkg_0.1/rtl/jtag_pkg.sv:8:11: Can't find typedef: 'jtag_req_t'
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_lc_ctrl_pkg_0.1/rtl/lc_ctrl_pkg.sv:40:11: '::' expected to reference a class/package but referenced '<unresolved-object>'

                       : ... Suggest '.' instead of '::'
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_lc_ctrl_pkg_0.1/rtl/lc_ctrl_pkg.sv:40:11: Can't find typedef: 'lc_tx_t'
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_jtag_pkg_0.1/rtl/jtag_pkg.sv:17:11: '::' expected to reference a class/package but referenced '<unresolved-object>'

                 : ... Suggest '.' instead of '::'
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_jtag_pkg_0.1/rtl/jtag_pkg.sv:17:11: Can't find typedef: 'jtag_rsp_t'
%Error: Internal Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_systems_flash_ctrl_pkg_0.1/rtl/autogen/flash_ctrl_pkg.sv:360:26: ../V3LinkDot.cpp:2210: Bad package link
```

This file also needs lc_ctrl_pkg.sv (see #495).

RRozak, 2021-10-01:
https://github.com/chipsalliance/UHDM-integration-tests/pull/525 fix the errors.
3 warnings left:
```
%Warning-WIDTH: Operator PATMEMBER expects 9 bits on the Pattern value, but Pattern value's CONST '65'h1' generates 65 bits.
                        : ... In instance prim_flash
                ... For warning description see https://verilator.org/warn/WIDTH?v=4.213
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: Operator PATMEMBER expects 9 bits on the Pattern value, but Pattern value's CONST '65'h2' generates 65 bits.
                        : ... In instance prim_flash
%Warning-WIDTH: Operator PATMEMBER expects 9 bits on the Pattern value, but Pattern value's CONST '65'h3' generates 65 bits.
                        : ... In instance prim_flash
```

RRozak, 2021-10-01:
I added rules for these warnings to .vlt file, because there were already some similar rules in it.

closed 2021-10-01 by RRozak