chipsalliance/UHDM-integration-tests #498: Earlgrey: bind_of_rv_and_dmdpi doesn't work in Verilator after Opentitan update

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, bind_of_rv_and_dmdpi test is failing with:
```
%Warning-WIDTH: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_ip_lc_ctrl_pkg_0.1/rtl/lc_ctrl_pkg.sv:177:60: Operator VAR 'TransTokenIdxMatrix' expects 1323 bits on the Initial value, but Initial value's REPLICATE generates 2987 bits.

                                : ... In instance top
                ... For warning description see https://verilator.org/warn/WIDTH?v=4.213
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

This file also needs lc_ctrl_pkg.sv (see #495).

RRozak, 2021-10-01:
Now it gives error
`%Error: Unknown object type: unsupported_typespec`

RRozak, 2021-10-01:
It uses prim_fifo_async (https://github.com/chipsalliance/UHDM-integration-tests/issues/494) that gives the same warning