chipsalliance/UHDM-integration-tests #500: Earlgrey: newest prim_otp doesn't work in Verilator

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, prim_otp.sv is failing with:
```
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:38:18: Parameter without initial value is never given value (IEEE 1800-2017 6.20.1): 'VendorTestOffset'

                 : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:39:18: Parameter without initial value is never given value (IEEE 1800-2017 6.20.1): 'VendorTestSize'

                 : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:85:25: Expecting expression to be constant, but variable isn't const: 'VendorTestOffset'

                 : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:78:23: Expecting expression to be constant, but variable isn't const: 'VendorTestSize'

                 : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:89:5: Can't convert defparam value to constant: Param 'VendorTestOffset' of 'u_impl_generic'

                : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_abstract_otp_0/prim_otp.sv:89:5: Can't convert defparam value to constant: Param 'VendorTestSize' of 'u_impl_generic'

                : ... In instance prim_otp
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_generic_otp_0/rtl/prim_generic_otp.sv:28:18: Parameter without initial value is never given value (IEEE 1800-2017 6.20.1): 'VendorTestOffset'

                            : ... In instance prim_otp.gen_generic.u_impl_generic
%Error: /media/hdd/uhdm/uhdm-integration/verilator/uhdm-integration/../uhdm-tests/opentitan/module_tests//..//opentitan/build/lowrisc_dv_chip_verilator_sim_0.1/src/lowrisc_prim_generic_otp_0/rtl/prim_generic_otp.sv:29:18: Parameter without initial value is never given value (IEEE 1800-2017 6.20.1): 'VendorTestSize'

                            : ... In instance prim_otp.gen_generic.u_impl_generic
%Error: Exiting due to 8 error(s)
```

RRozak, 2021-09-30:
The same errors were thrown by original verilator.
In #520 I added missing modules and initialized the parameters.
Now there are 2 warnings:
```
lc_ctrl_pkg.sv:177:60: Operator VAR 'TransTokenIdxMatrix' expects 1323 bits on the Initial value, but Initial value's REPLICATE generates 2987 bits.
prim_generic_otp.sv:367:3: Input port connection 'wmask_i' expects 22 bits on the pin connection, but pin connection's REPLICATE generates 28 bits.
```
The first is from package that doesn't work (https://github.com/chipsalliance/UHDM-integration-tests/issues/495) and occurs in other modules too. I will wait with it until it is solved.
I will start investigating the second now.

RRozak, 2021-09-30:
The second warning should be fixed by https://github.com/chipsalliance/Surelog/issues/1977

RRozak, 2021-09-30:
The warning can be reproduced by https://github.com/chipsalliance/UHDM-integration-tests/pull/521

RRozak, 2021-10-01:
It works after https://github.com/antmicro/verilator/pull/508

closed 2021-10-01 by RRozak