chipsalliance/UHDM-integration-tests #512: Earlgrey: alert_handler_reg_wrap doesn't work in Verilator after Opentitan update

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, alert_handler_reg_wrap test is failing with:
```
%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:346: Null item passed to setOp1p
```
Looks like we're missing some dependencies.

RRozak, 2021-09-30:
https://github.com/chipsalliance/UHDM-integration-tests/pull/520 adds missing dependencies.
But there are another errors, which are thrown only when uhdm-verilator is used:
```
prim_subreg.sv:12:39: Too many digits for 16 bit number: 16'd18446744073709551583
prim_subreg.sv:12:39: Too many digits for 2 bit number: 2'd18446744073709551615
prim_subreg.sv:12:39: Too many digits for 2 bit number: 2'd18446744073709551614
prim_subreg.sv:12:39: Too many digits for 2 bit number: 2'd18446744073709551613
prim_subreg.sv:12:39: Too many digits for 2 bit number: 2'd18446744073709551612
prim_subreg.sv:12:39: Too many digits for 16 bit number: 16'd18446744073709551615
prim_subreg.sv:12:39: Too many digits for 32 bit number: 32'd18446744073709551615
```

RRozak, 2021-09-30:
The errors can be reproduced by https://github.com/chipsalliance/UHDM-integration-tests/pull/522
It is blocked by https://github.com/chipsalliance/Surelog/issues/1978