chipsalliance/UHDM-integration-tests #513: Earlgrey: csrng_core doesn't work in Verilator after Opentitan update

rkapuscik, 2021-09-29:
After updating Opentitan to newest version, csrng_core test is failing with:
```
%Error: Could not get lowconn handle for port, aborting
%Error: Internal Error: ../V3Ast.cpp:346: Null item passed to setOp1p    
```
Looks like we're missing some dependencies.

RRozak, 2021-09-30:
#520 adds missing dependencies.
After it, there is the warning thrown:
`lc_ctrl_pkg.sv:177:60: Operator VAR 'TransTokenIdxMatrix' expects 1323 bits on the Initial value, but Initial value's REPLICATE generates 2987 bits.`
There is the issue about that package: https://github.com/chipsalliance/UHDM-integration-tests/issues/495, so I mark it as blocked and come back after fixing the problems with that package.

RRozak, 2021-10-01:
It works after https://github.com/antmicro/verilator/pull/508

closed 2021-10-01 by RRozak