chipsalliance/UHDM-integration-tests #598: SERV get miscompiled

olofk, 2021-11-19:
SERV has a line that looks like this `wire co_ctrl_utype       = !opcode[4] & opcode[2] & opcode[0];`. When opcode is 0x5 the wire should be asserted, but when I compile with Surelog it remains low. If I remove related code that reads the result signal, it behaves as expected again.

Attaching a minimal test case using surelog+uhdm and a reference build using verilator directly + a gtkwave screenshot showing the difference
[surelog_bug2.tar.gz](https://github.com/chipsalliance/UHDM-integration-tests/files/7570441/surelog_bug2.tar.gz)

[image]

tgorochowik, 2021-11-19:
Thanks! We will take a look at this!

rkapuscik, 2021-11-26:
Should be fixed by https://github.com/antmicro/verilator/pull/592. I also added your test case in https://github.com/chipsalliance/UHDM-integration-tests/pull/599, but tested this only by verifying that the waveforms are the same between UHDM and original Verilator.

Please check if it's working now for you and report other issues if you encounter any.

olofk, 2021-11-26:
Yes! Tested and works fine. Thanks for this. Got it running with a full SERV SoC via FuseSoC+Edalize using the wip [surelog](https://github.com/olofk/edalize/tree/surelog) branch

closed 2021-11-26 by olofk